pymodbus-dev/pymodbus #2640: Ability to override response class for specific request.

yawor, 2025-04-15:
Hi,

I've stumbled upon a device, which doesn't follow the modbus specification properly. It's a Waveshare 8-channel relay module. They implemented it in such a way, that modbus function 5 (write single coil) can not only accept 0xFF (ON) or 0x00 (OFF) as a value, but also, depending on specific address range, it accepts also 0x55 (which toggles state of selected coil) or even accepts almost any 16-bit value which represents a time for which the coil gets turned on (in multiples of 100 ms).

I could register a custom `ModbusPDU` with `function_code = 5` but it would replace the `WriteSingleCoilResponse` for the whole program, and I still have other devices on the bus, so I don't think that replacing the response class for that code globally is a way to go.

I've been thinking about the ability to override the response lookup for a single request, by providing optional `response_class` or `response_pdu` argument to `execute` method in the client. It would then set a ContextVar defined maybe in decoders.py, which then be checked in the `DecodePDU.lookupPduClass` method if it is set and if it contains a class for received function code. If both conditions are not met, then a normal lookup would follow.

The ContextVar would be reset before the `execute` method returns (with the use of try/finally).

I can prepare a PR with this if there are no objections related to the functionality.

janiversen, 2025-04-15:
You do not need to modify modify pymodbus to achieve what you want!

Simple add a trace_pdu or trace_packet funcion. in the funcion check for device and modify the packet/pdu.

A custom_pdu could do exactly the same, inherit the original class and if not you device just call super(). I do that for a device I have.

janiversen, 2025-04-15:
ps. custom_pdu is one of the lesser tested parts of pymodbus, so if you use that and run into problems please let me know.

janiversen, 2025-04-15:
btw rereading your text, it seems you need to overwrite the request PDU, not the response PDU ??

if so, that is done easily by making a custom request PDU, and calling execute with that. The standard WriteCoil response do not need to be modified because if the data type != 0x00 then true is returned.

yawor, 2025-04-15:
@janiversen request PDU is not a problem. I do that already and I can see that the relay is working as it should. I get back the `WriteSingleCoilResponse` as you mentioned, but I loose the response details, as the returned data simply gets converted to `True`. In this specific case this may not be that much important, as the module basically responds with exact data that has been sent in the request.

Regarding trace_pdu and trace_packet, if I understand correctly:
- trace_packet allows to watch and modify byte stream received from the transport before decoding or being sent to the transport after encoding
- trace_pdu allows to watch and modify the PDUs after decoding or before encoding.

Given that, if I still wanted to have a proper response PDU, I see two options:
1. Use trace_packet to detect data coming from specific device with function code 5 and specific address range by inspecting first four bytes and if it matches, modify the function code to something else and then register a response PDU to that custom function code. The issue here is that changing the response data would cause the CRC to be incorrect when the data is passed to the decoder. I would need to check the CRC here myself first to be sure it was correct in the first place and then change the data and regenerate new CRC before passing it to the decoder.
2. Use trace_pdu, but here I already get the decoded PDU that is already of specific class.

If I would like to publish my code for controlling the device as a library dependant on pymodbus, then there's a risk that some user uses that device and some other devices on the same bus and if the other devices also improperly reuse the same function codes then there may be a lot of issues when multiple libraries try to do some hacking to handle responses.

What I propose is a standard way to override possible response PDUs explicitly for specific execute without cluttering the global function code to PDU class mapping, which would require no additional logic in the application and with a minimal changes to the pymodbus which are fully backward compatible.

janiversen, 2025-04-15:
for 1) pdu contains a function to calculate the crc, so its a simple call.

The standard way to override an existing response PDU is to make a new class the inherits from the original, and in that class do what needs to be done for the specific device, and if not call super().func to allow the standard handling for all other devices. With that solution you have a specific response for your device, and the normal response for all others.

There are no need to change in pymodbus to do that.

janiversen, 2025-04-15:
2 different libraries would use 2 different instances of the ModbusClient object, and thus the registrations would not interfere (as they are done in the object, not the class)

If an app would need 2 different ResponsePDU for the same function code, the app could simple merge these or have one inherit from the other.

Feel free to suggest the change, the problem is not being "backward compatible", the problem is documentation and future maintenance. A pull request along the lines you described will not be accepted (unless of course the PR solve an issue, which cannot be solved otherwise).

yawor, 2025-04-15:
Can there be multiple ModbusClient objects for the same serial interface?

janiversen, 2025-04-15:
No, because that would create a conflict on the line (remember client is the master).

This means in essence that 2 libraries cannot share the same rs485 line and thus not the same ModbusClient objects.

yawor, 2025-04-15:
The library for some hardware doesn't need to create its own client instance. It can be passed the client instance created outside of the library. That way multiple libraries can actually use the same ModbusClient object and the same serial line. The TransactionManager uses locks to prevent multiple active requests at the same time so if there are multiple requests at the same time, then at most some of them would need to wait for active one to finish first before the next one is actually executed.

Anyway, as an author/maintainer of course you have the last word in how you envision your library, so I'll just back off as I see there's not much chance even if I take my time and create the fully working PR to get this merged.

closed 2025-04-15 by yawor

janiversen, 2025-04-15:
you are in theory right, but have you ever seen a library that did that.....normally the library controls the parameters of the rs485.

Good luck in your continued endeavor, if you find other items in pymodbus feel free to open a new issue.