pymodbus-dev/pymodbus #2646: Can not use client.convert_from_registers()

Edgar-qiqi, 2025-04-22:
After installing pymodbus version 3.9.2 via pip and importing from pymodbus import payload, client, the method client.convert_from_registers() remains inaccessible. Additionally, despite documentation stating that pymodbus.logging:BinaryPayloadDecoder was deprecated post-3.9.0, it persists in version 3.9.2. Furthermore, the removal of client.convert_from_registers() conflicts with earlier API specifications. This indicates potential inconsistencies between versioning claims, API documentation, and actual implementation in the 3.9.x release branch.

[image]

Edgar-qiqi, 2025-04-22:
[image]

janiversen, 2025-04-22:
We postponed the removal of payload, and honestly you are the first to complain about that. 

You do not share your code, so I cannot tell you why your app cannot find convert_from_registers. But I can tell you that if convert_from_registers was gone a lot of the test harness would fail (including a larger number of production systems).

Did you follow our examples ??

you write "importing from pymodbus import payload, client", I assume you mean pymodbus.client, but that do not have convert_from_registers, that is part of the client classes. I think you need to look at the examples or the documentation.

janiversen, 2025-04-23:
Closing as this is not a bug in pymodbus, but a wrong usage.

closed 2025-04-23 by janiversen