pymodbus-dev/pymodbus #2647: ModbusServerContext hardcoded value of 0xF7 used for __delitem__ func

sush2023, 2025-04-22:
Currently using version 3.8.0, same behavior for version ~3.9

The __delitem__ func ensures the slave parameter is between 0xF7 and 0 before deleting the key/node. While I am able to have a slave at an address higher with proper functionality, deleting is not working due to that code throwing an exception.

Line of interest:
https://github.com/pymodbus-dev/pymodbus/blob/28eb3194692a2001bd966459892199133e84cbee/pymodbus/datastore/context.py#L203

janiversen, 2025-04-23:
pull requests are welcome.

however I am not sure why you would first add a device and then later delete it. If you delete a device in a running in a running server you will sooner or later have a server breakdown.

sush2023, 2025-04-24:
Ideally it would be best to be able to have the device not respond at a certain address. Removing seemed to be the most plausible option.

janiversen, 2025-04-25:
Closing as this is not a pymodbus bug.

closed 2025-04-29 by janiversen