pymodbus-dev/pymodbus #2650: Simulator datatypes issue

dakhnod, 2025-04-29:
Hello, thanks for the work on this!

I tried searching through the configuration, surely I missed something.

When starting the simulator, it already has some registers pre-filled, some not.

When attempting to write a value to a register with type `invalid`, I get the error `Illegal data address`.

I do not quite understand which addresses are restricted to which types and how to change a registers type.

I would be thankful for any input!

Thanks again!

janiversen, 2025-04-29:
Please read the documentation, you need to supply a configuration, there is nothing predefined.

janiversen, 2025-04-29:
please read the documentation!!

'just running' yields a default simulator or one configured for your needs.

janiversen, 2025-04-29:
Does not seem to be a bug in pymodbus, but more a matter of reading the documentation (e.g. explore the configuration)