pymodbus-dev/pymodbus #2661: Python version issue during docker build.(3.12.7-r0)

sso9409, 2025-05-04:
* This post was written using a translator.

Hello, I am using your repository to post what I solved while experiencing an error during the docker build myself.

The error appears to be a Python version collision.

This log during build...
``` log
.
.
.
docker.io/library/alpine:3.20.3@sha256:1e42bbe2508154c9126d48c2b8a75420c3544343bf86fd041fb7527e017a4b4a              0.1s
 => ERROR [server1 2/4] RUN apk upgrade --available --no-cache --update     && apk add --no-cache --update        python3=3.12.7-  11.1s 
------
 > [server1 2/4] RUN apk upgrade --available --no-cache --update     && apk add --no-cache --update        python3=3.12.7-r0        py3-pip=24.0-r2     && rm -rf /var/cache/apk/* /tmp/* /var/tmp/*:
0.698 fetch https://dl-cdn.alpinelinux.org/alpine/v3.20/main/x86_64/APKINDEX.tar.gz
1.414 fetch https://dl-cdn.alpinelinux.org/alpine/v3.20/community/x86_64/APKINDEX.tar.gz
2.997 Upgrading critical system libraries and apk-tools:
2.997 (1/1) Upgrading apk-tools (2.14.4-r0 -> 2.14.4-r1)
3.077 Executing busybox-1.36.1-r29.trigger
3.135 Continuing the upgrade transaction with new apk-tools:
3.139 fetch https://dl-cdn.alpinelinux.org/alpine/v3.20/main/x86_64/APKINDEX.tar.gz
3.609 fetch https://dl-cdn.alpinelinux.org/alpine/v3.20/community/x86_64/APKINDEX.tar.gz
6.039 (1/5) Upgrading musl (1.2.5-r0 -> 1.2.5-r1)
6.407 (2/5) Upgrading ca-certificates-bundle (20240705-r0 -> 20241121-r1)
6.531 (3/5) Upgrading libcrypto3 (3.3.2-r0 -> 3.3.3-r0)
7.740 (4/5) Upgrading libssl3 (3.3.2-r0 -> 3.3.3-r0)
7.996 (5/5) Upgrading musl-utils (1.2.5-r0 -> 1.2.5-r1)
8.033 Executing busybox-1.36.1-r29.trigger
8.039 OK: 8 MiB in 14 packages
8.092 fetch https://dl-cdn.alpinelinux.org/alpine/v3.20/main/x86_64/APKINDEX.tar.gz
8.674 fetch https://dl-cdn.alpinelinux.org/alpine/v3.20/community/x86_64/APKINDEX.tar.gz
10.04 ERROR: unable to select packages:
8.674 fetch https://dl-cdn.alpinelinux.org/alpine/v3.20/community/x86_64/APKINDEX.tar.gz
10.04 ERROR: unable to select packages:
10.04 ERROR: unable to select packages:
10.04   python3-3.12.10-r0:
10.04   python3-3.12.10-r0:
10.04     breaks: world[python3=3.12.7-r0]
10.04     satisfies: py3-parsing-pyc-3.1.2-r1[python3]
10.04                py3-parsing-pyc-3.1.2-r1[python3~3.12]
10.04                py3-pip-24.0-r2[python3]
10.04                py3-pip-24.0-r2[python3~3.12]
10.04                py3-parsing-pyc-3.1.2-r1[python3~3.12]
10.04                py3-pip-24.0-r2[python3]
10.04                py3-pip-24.0-r2[python3~3.12]
10.04                py3-setuptools-70.3.0-r0[python3]
10.04                py3-setuptools-70.3.0-r0[python3~3.12]
10.04                py3-parsing-3.1.2-r1[python3]
10.04                py3-parsing-3.1.2-r1[python3~3.12]
10.04                py3-parsing-pyc-3.1.2-r1[python3~3.12]
10.04                py3-pip-24.0-r2[python3]
10.04                py3-pip-24.0-r2[python3~3.12]
10.04                py3-setuptools-70.3.0-r0[python3]
10.04                py3-setuptools-70.3.0-r0[python3~3.12]
10.04                py3-parsing-3.1.2-r1[python3]
10.04                py3-parsing-pyc-3.1.2-r1[python3~3.12]
10.04                py3-pip-24.0-r2[python3]
10.04                py3-pip-24.0-r2[python3~3.12]
10.04                py3-setuptools-70.3.0-r0[python3]
10.04                py3-setuptools-70.3.0-r0[python3~3.12]
10.04                py3-parsing-pyc-3.1.2-r1[python3~3.12]
10.04                py3-pip-24.0-r2[python3]
10.04                py3-pip-24.0-r2[python3~3.12]
10.04                py3-setuptools-70.3.0-r0[python3]
10.04                py3-parsing-pyc-3.1.2-r1[python3~3.12]
10.04                py3-pip-24.0-r2[python3]
10.04                py3-pip-24.0-r2[python3~3.12]
10.04                py3-setuptools-70.3.0-r0[python3]
10.04                py3-setuptools-70.3.0-r0[python3~3.12]
10.04                py3-pip-24.0-r2[python3]
10.04                py3-pip-24.0-r2[python3~3.12]
10.04                py3-setuptools-70.3.0-r0[python3]
10.04                py3-setuptools-70.3.0-r0[python3~3.12]
10.04                py3-parsing-3.1.2-r1[python3]
10.04                py3-parsing-3.1.2-r1[python3~3.12]
10.04                py3-setuptools-70.3.0-r0[python3]
10.04                py3-setuptools-70.3.0-r0[python3~3.12]
10.04                py3-parsing-3.1.2-r1[python3]
10.04                py3-parsing-3.1.2-r1[python3~3.12]
10.04                py3-setuptools-70.3.0-r0[python3~3.12]
10.04                py3-parsing-3.1.2-r1[python3]
10.04                py3-parsing-3.1.2-r1[python3~3.12]
10.04                py3-packaging-24.0-r1[python3]
10.04                py3-parsing-3.1.2-r1[python3]
10.04                py3-parsing-3.1.2-r1[python3~3.12]
10.04                py3-packaging-24.0-r1[python3]
10.04                py3-packaging-24.0-r1[python3]
10.04                py3-packaging-24.0-r1[python3~3.12]
10.04                py3-packaging-pyc-24.0-r1[python3]
10.04                py3-packaging-24.0-r1[python3~3.12]
10.04                py3-packaging-pyc-24.0-r1[python3]
10.04                py3-packaging-pyc-24.0-r1[python3]
10.04                py3-packaging-pyc-24.0-r1[python3~3.12]
------
time="2025-05-04T09:25:58+09:00" level=warning msg="current commit information was not captured by the build" error="failed to read current commit information with git rev-parse --is-inside-work-tree"
failed to solve: process "/bin/sh -c apk upgrade --available --no-cache --update     && apk add --no-cache --update        python3=3.12.7-r0        py3-pip=24.0-r2     && rm -rf /var/cache/apk/* /tmp/* /var/tmp/*" did not complete successfully: exit code: 7
```

So I corrected it in the docker file like this and solved it.

[image]

I don't know if this project is managed right now, but I'm uploading it to help anyone.

Thank you.

sso9409, 2025-05-04:
I'm sorry. I uploaded it wrong.

janiversen, 2025-05-04:
??? then why is the issue still open ?

sso9409, 2025-05-04:
I'm sorry. I don't know how to change the status of the issue.

closed 2025-05-04 by sso9409