pymodbus-dev/pymodbus #2668: Decode 8 bit not possible

hasenradball, 2025-05-14:
Dear pymodbus team,

with version 3.6.6 it was possible to decode 8 bit values.
But with version 3.9.2 this is no possible anymore, because the `DATATYPE.UINT8` and `DATATYPE.INT8` are missing.

Could you provide also these datatypes for backward compatiliby?

I used the following function:

```
def decode_register_readings(self, readings, datatype, count) -> list:
        """Decode the register readings 
        
        Decode readings depending on datatype given
        -----
        Args:
            readings: dats read from register
            datatype: U16, U32 , etc...
            count: number of datatypes to be converted
        
        Returns:
            data: list of decoded values
        """
        decoder = BinaryPayloadDecoder.fromRegisters(readings.registers, \
            byteorder=Endian.BIG, wordorder=Endian.BIG)
        #print(f'decoder : {decoder}')
        data = []
        if datatype == 'U8':
            data = [decoder.decode_8bit_uint() for i in range(count)]
        elif datatype == 'U16':
            data = [decoder.decode_16bit_uint() for i in range(count)]
        elif datatype == 'U32':
            data = [decoder.decode_32bit_uint() for i in range(count)]
        elif datatype == 'U64':
            data = [decoder.decode_64bit_uint() for i in range(count)]
        elif datatype == 'S8':
            data = [decoder.decode_8bit_int() for i in range(count)]
        elif datatype == 'S16':
            data = [decoder.decode_16bit_int() for i in range(count)]
        elif datatype == 'S32':
            data = [decoder.decode_32bit_int() for i in range(count)]
        elif datatype == 'S64':
            data = [decoder.decode_64bit_int() for i in range(count)]
        return data
```

janiversen, 2025-05-14:
that still have nothing to do with missing entries in DATATYPE.

We will not add INT8 and UINT8 as a modbus device cannot return an 8 bit values, so if you need that you have to decode it in the app.

As I wrote payload will be removed in v4.0 we postponed it from v3.9.0, so you have time.

janiversen, 2025-05-14:
How can a modbus device return 8bit, that is really non standard. Modbus devices always return registers which are defined to be 16bit!

janiversen, 2025-05-14:
If you receive a register (16bit) from your device via pymodbus, and for any reason want to split it into 2 bytes, then use the builtin python functions, it really is outside the scope of pymodbus.

janiversen, 2025-05-14:
Closing as this is not a pymodbus problem.

hasenradball, 2025-05-14:
Ok Thank you for the feedback.