pymodbus-dev/pymodbus #2677: Question about memory fragmentation on old version 2.5.x.

ieb, 2025-06-01:
I have a python service using 2.5.0 performing serial ModbusRTU in a VictronEnergy Multiplus II inverter. All other python servicse using mostly the same stack except ModbusRTU run continuously with a stable RSS and VSZ  (eg 15MB/50MB), however the service that uses pymodbus grows in size reaching 90MB/124MB  after 24h having started at 23MB/44MB and continues to grow. I have verified through gc, _tracemalloc, heapy that there there are no leaks in python objects or variables and gc is operating as expected.  I have a core dump but unfortunately no dbg symbols are available to analyse it and find the cause, so I am bisecting and reading code.  Obviously I can restart every 24h, but I would rather not leak. Python is 3.8 and hard if not impossible to change as the OS is VenusOS and commercial. 

Is it possible that this copy operation would cause python memory fragmentation and cause constant growth in RSS without showing up in Python ?  Or is this just the way it will be with Python 3.8 and fragmentation. IRC Python 2.x wasnt great.

https://github.com/pymodbus-dev/pymodbus/blob/c5772b35ae3f29d1947f3ab453d8d00df846459f/pymodbus/framer/rtu_framer.py#L106

If it is likely, I can try and patch. Just looking for expert knowledge.

For reference https://github.com/victronenergy/venus/issues/1465

closed 2025-06-01 by janiversen

ieb, 2025-06-02:
Thanks, will give 3.9.2 another go.

janiversen, 2025-06-11:
Are you saying the problem still exist in 3.9.2 Quote "No version of PyModbus or PySerial eliminated the problem"

And since state "the leak was in" it seems you have solved it, then why not make a PR and let others benefit from it.

ieb, 2025-06-12:
> If it's a problem between the USB hardware and the ch341 kernel module, then it would be more correct to state that pymodbus do not have a problem.

This  ^^^ is the correct statement. As far as I can tell, pymodbus does not have a memory leak problem in any version.

I was not able to eliminate the leak in the process, but I was able to reduce it by aggressively backing of retries to timeout conditions, hence no PR to share to this or other projects.