pymodbus-dev/pymodbus #2688: Even parity doesn´t work with AsyncModbusSerialClient

Kampi, 2025-07-16:
I use the following code

```
import sys
import asyncio
import logging
import time

import pymodbus.client as modbusClient

_logger = logging.getLogger(__name__)
logging.basicConfig(level=logging.DEBUG)

async def main():
    client = modbusClient.AsyncModbusSerialClient(
            "COM16",
            # Common optional parameters:
            timeout = 1,
            #    retries=3,
            baudrate = 19200,
            framer = "rtu",
            bytesize = 8,
            parity = "E",
            stopbits = 1,
            #    handle_local_echo=False,
        )

    _logger.info("Client starting")
    await client.connect()
    assert client.connected

    rr = await client.read_coils(0, count = 1, slave = 0x21)
    rr = await client.read_holding_registers(0, count = 1, slave = 0x21)
    print(rr.registers)

    client.close()
    _logger.info("End of Program")

if(__name__ == "__main__"):
    asyncio.run(main(), debug = True)
```

`read_coils` transmit the bytes `0x21 0xe0 0x1 0xa9 0xca` but the logic analyzer shows an invalid transmission

[image]

Replacing `even` parity with `none` fixes the issue

[image]

janiversen, 2025-07-17:
What did you expect when using Even ? are you sure your uart support 8bit and parity ?

Pymodbus is just passing the parity parameter to the UART, so I am not sure how pymodbus can do something wrong.

However I will not exclude that it is a bug.

Kampi, 2025-07-19:
Interesting. It's a regular FT232 adapter. But thanks for the response. I will close the ticket. :)