pymodbus-dev/pymodbus #2698: Exception handling missing for malformed frames (e.g. struct.error)

dmc4, 2025-07-26:
Dear Pymodbus developers: 

Hope you're doing well, and thank you for your continued work on the project.

While integrating the latest version of `pymodbus`, we encountered a number of unhandled exceptions related to malformed or truncated Modbus frames. These exceptions seem to arise from missing defensive checks during frame decoding.

---

### 📌 Problem Description

In multiple cases (at least 8 confirmed), malformed Modbus payloads trigger a fatal `struct.error` due to insufficient length checks before unpacking data. Here's a representative traceback:

```text
ERROR:asyncio:Fatal error: protocol.data_received() call failed.
protocol: <pymodbus.server.requesthandler.ServerRequestHandler object at 0x71cff25c13a0>
transport: <_SelectorSocketTransport fd=7 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File ".../asyncio/selector_events.py", line 1027, in _read_ready__data_received
    self._protocol.data_received(data)
  File ".../pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
  File ".../pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File ".../pymodbus/server/requesthandler.py", line 79, in callback_data
    used_len = super().callback_data(data, addr)
  File ".../pymodbus/transaction/transaction.py", line 217, in callback_data
    used_len, pdu = self.framer.processIncomingFrame(self.trace_packet(False, data))
  File ".../pymodbus/framer/base.py", line 76, in processIncomingFrame
    data_len, pdu = self._processIncomingFrame(data[used_len:])
  File ".../pymodbus/framer/base.py", line 97, in _processIncomingFrame
    if (result := self.decoder.decode(frame_data)) is None:
  File ".../pymodbus/pdu/decoders.py", line 112, in decode
    pdu.decode(frame[1:])
  File ".../pymodbus/pdu/register_message.py", line 180, in decode
    self.address, register = struct.unpack(">HH", data)
struct.error: unpack requires a buffer of 4 bytes
```

---

### 🔬 Example Payloads

Below is an example payload that can trigger this behavior:

```
68e80000000201022270967f
```

If needed, we can provide the full set of payloads that led to these errors.

---

### 📎 Environment

* `pymodbus` version: 3.9.2
* Python: 3.12
* Platform: Linux (Ubuntu)
* Server mode: TCP
* Customizations: none

---

Thanks again for your great work on `pymodbus` — we really appreciate the flexibility it provides. Please let us know if a patch or test case would help; we're happy to contribute if needed.

Best regards,
Changze

dmc4, 2025-07-27:
Thanks for the quick replies, Jan.

To clarify: all the frames I’m feeding into pymodbus are intentionally malformed, generated by a custom fuzzer I developed. The goal is to test the robustness of the frame decoding logic — that is, ensuring pymodbus can gracefully handle invalid or truncated frames without raising fatal exceptions like struct.error.

I fully understand that the payloads are not valid Modbus frames — I’m not expecting them to be processed successfully. But I believe pymodbus should fail safely, i.e. detect bad input and reject it cleanly, rather than crash with uncaught exceptions.

If it is still helpful, I’ll follow up with:
Full debug logs showing the 8 crashing cases

Let me know what would be most helpful.

Best,
Changze

janiversen, 2025-07-27:
Pymodbus cannot possibly react well to all types of malicious frames it receives, the important point is that it do not deliver false information to the app.

The documentation states that pymodbus work with standard modbus, not deviations, so it is not a goal per se, to handle all types of malicious frames.

Having said that, the more graceful pymodbus handle, in this case, malicious frames, the better. So please if you want better handling feel free to submit a pull request.

Ps. It is actually very easy to avoid the reported crash, but it might have some ripple effects.

closed 2025-07-27 by dmc4

henzef, 2025-09-24:
Hi @dmc4,
did you find the time to improve the robustness of pymodbus in this regard?
I think this is related to [2556](https://github.com/pymodbus-dev/pymodbus/issues/2556), which is an issue for me as well.

Regards