pymodbus-dev/pymodbus #2777: Transaction ID mismatch

Simon-SGB, 2025-09-16:
Using pymodbus (3.11.2) in a TCP client application works quiet well for me. I use the ModBusTCPClient Synchronous API and have no issues when on the local network.

As soon as I have my client behind an LTE gateway with NAT, I sometimes see mismatching transaction IDs:

```
...
2025-09-16 16:05:55,539 DEBUG base:72 Processing: 0x0 0x1b 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x0
2025-09-16 16:05:55,539 DEBUG decoders:79 decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[0], status=1, retries=0)
2025-09-16 16:05:55,696 DEBUG base:72 Processing: 0x0 0x1b 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x0 0x0 0x1c 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2
2025-09-16 16:05:55,696 ERROR base:83 ERROR: request ask for transaction_id=28 but got id=27, Skipping.
2025-09-16 16:05:55,696 DEBUG base:72 Processing: 0x0 0x1b 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x0 0x0 0x1c 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2
2025-09-16 16:05:55,696 ERROR base:83 ERROR: request ask for transaction_id=28 but got id=27, Skipping.
2025-09-16 16:05:58,911 DEBUG base:72 Processing: 0x0 0x1c 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2
2025-09-16 16:05:58,911 DEBUG decoders:79 decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[2], status=1, retries=0)
...
```

This still kind of works, but I would like to understand what happens. Has anyone observed something like this?

janiversen, 2025-09-16:
You only show a very limited log, without the send, so it is impossible to tell you why this happens.

NAT itself should not have anything to do with transaction id, except if some of the chain duplicates the records.

Please activate debug, with the builtin log call, and present a log where we can see a couple of requests (send) and how they are replied.

please make it first without NAT (where it works) and then with NAT, enabling us to see the differences.

Simon-SGB, 2025-09-17:
After connecting the `ModbusTcpClient`, I already have this line in my code:
```
logging.pymodbus_apply_logging_config(level="Debug")
```
I am still seeing no requests, is there anything else required?

Simon-SGB, 2025-09-17:
I just tried to use `AsyncModbusTcpClient` instead. This reproduces the same issue, but the stdout does include output from `send` and `recv`. Here, the output with some more context:

```
2025-09-17 15:43:33,921 DEBUG transport:375 send: 0x0 0x12 0x0 0x0 0x0 0x6 0x0 0x4 0x1 0x35 0x0 0x1
2025-09-17 15:43:34,089 DEBUG transport:329 recv: 0x0 0x12 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x0 extra data:
2025-09-17 15:43:34,089 DEBUG base:72 Processing: 0x0 0x12 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x0
2025-09-17 15:43:34,091 DEBUG decoders:79 decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[0], status=1, r
etries=0)
2025-09-17 15:43:34,091 DEBUG transport:375 send: 0x0 0x13 0x0 0x0 0x0 0x6 0x0 0x4 0x1 0x36 0x0 0x2
2025-09-17 15:43:34,261 DEBUG transport:329 recv: 0x0 0x13 0x0 0x0 0x0 0x7 0x0 0x4 0x4 0x0 0xf 0x42 0x3f extra data:
2025-09-17 15:43:34,261 DEBUG base:72 Processing: 0x0 0x13 0x0 0x0 0x0 0x7 0x0 0x4 0x4 0x0 0xf 0x42 0x3f
2025-09-17 15:43:34,263 DEBUG decoders:79 decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[15, 16959], sta
tus=1, retries=0)
2025-09-17 15:43:34,263 DEBUG transport:375 send: 0x0 0x14 0x0 0x0 0x0 0x6 0x0 0x4 0x1 0x38 0x0 0x1
2025-09-17 15:43:34,431 DEBUG transport:329 recv: 0x0 0x14 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2 extra data:
2025-09-17 15:43:34,432 DEBUG base:72 Processing: 0x0 0x14 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2
2025-09-17 15:43:34,432 DEBUG decoders:79 decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[2], status=1, r
etries=0)
2025-09-17 15:43:34,432 DEBUG transport:375 send: 0x0 0x15 0x0 0x0 0x0 0x6 0x0 0x4 0x1 0x39 0x0 0x1
2025-09-17 15:43:34,601 DEBUG transport:329 recv: 0x0 0x15 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x1 extra data:
2025-09-17 15:43:34,601 DEBUG base:72 Processing: 0x0 0x15 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x1
2025-09-17 15:43:34,601 DEBUG decoders:79 decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[1], status=1, r
etries=0)
2025-09-17 15:43:34,606 DEBUG transport:375 send: 0x0 0x16 0x0 0x0 0x0 0x6 0x0 0x4 0x1 0x3a 0x0 0x1
2025-09-17 15:43:37,610 DEBUG transport:375 Repeating....
2025-09-17 15:43:38,921 DEBUG transport:329 recv: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2 extra data:
2025-09-17 15:43:38,921 DEBUG base:72 Processing: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2
2025-09-17 15:43:38,923 DEBUG decoders:79 decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[2], status=1, r
etries=0)
2025-09-17 15:43:38,923 DEBUG transport:375 send: 0x0 0x17 0x0 0x0 0x0 0x6 0x0 0x4 0x1 0x3b 0x0 0x2
2025-09-17 15:43:39,082 DEBUG transport:329 recv: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2 0x0 0x17 0x0 0x0 0x0 0x7 0x0 0x4 0x4 0xff 0xff 0xfe 0x47 extra data:
2025-09-17 15:43:39,083 DEBUG base:72 Processing: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2 0x0 0x17 0x0 0x0 0x0 0x7 0x0 0x4 0x4 0xff 0xff 0xfe 0x47
2025-09-17 15:43:39,085 ERROR base:83 ERROR: request ask for transaction_id=23 but got id=22, Skipping.
2025-09-17 15:43:39,085 DEBUG base:72 Processing: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2 0x0 0x17 0x0 0x0 0x0 0x7 0x0 0x4 0x4 0xff 0xff 0xfe 0x47
2025-09-17 15:43:39,085 ERROR base:83 ERROR: request ask for transaction_id=23 but got id=22, Skipping.
2025-09-17 15:43:39,085 DEBUG base:72 Processing: 0x0 0x16 0x0 0x0 0x0 0x5 0x0 0x4 0x2 0x0 0x2 0x0 0x17 0x0 0x0 0x0 0x7 0x0 0x4 0x4 0xff 0xff 0xfe 0x47
2025-09-17 15:43:39,085 ERROR base:83 ERROR: request ask for transaction_id=23 but got id=22, Skipping.
2025-09-17 15:43:41,938 DEBUG transport:375 send: 0x0 0x17 0x0 0x0 0x0 0x6 0x0 0x4 0x1 0x3b 0x0 0x2
2025-09-17 15:43:42,101 DEBUG transport:329 recv: 0x0 0x17 0x0 0x0 0x0 0x7 0x0 0x4 0x4 0xff 0xff 0xfe 0x47 extra data:
2025-09-17 15:43:42,101 DEBUG base:72 Processing: 0x0 0x17 0x0 0x0 0x0 0x7 0x0 0x4 0x4 0xff 0xff 0xfe 0x47
2025-09-17 15:43:42,101 DEBUG decoders:79 decoded PDU function_code(4 sub -1) -> ReadInputRegistersResponse(dev_id=0, transaction_id=0, address=0, count=0, bits=[], registers=[65535, 65095],
status=1, retries=0)
```

janiversen, 2025-09-17:
It seems your network have some serious problems.

at 34,606 a request is sent with tid=22. For some strange reason it is repeated at 34,610 this only happens if socket.send reports a send failure, in which case it must be repeated.

at 38,921  a response with tid=22 is received and handled correctly. This closes the handling of tid=22

at 39.923 a new request is sent with tid=23

But then at 39,083 and 39,085 the response with tid=22 is received. Due to the send failure, it would be understandable if it was received 2 times, but it is received 3 times, this is very unlikely a duplication from your device. HOWEVER this is ignored, so it does not affect your program.

at 41,938 the request with tid=23 is repeated due to timeout.

at 42,101 the response with tid=23 received.

As you can see, there are some instabilities in your network, it can be caused by the NAT software, however normally NAT only affect the connect stage.

But as you can also see, the failures do not affect your program, and I cannot see any bugs in pymodbus.

closed 2025-09-17 by Simon-SGB