pymodbus-dev/pymodbus #2779: ModbusSlaveContext, removed zero_mode parameter.

bambosd, 2025-09-24:
Hello everyone.

2 releases ago the mode zero for modbus tcp server is removed.
so zero based mapping is not possible. (request for register 0 replies with the value on register 1).

Is there any way to fix this ?

janiversen, 2025-09-24:
It is not 2 releases ago, it is more like circa 2 years ago.

Since requests for register 0 replies with the definition 1, and register 1 replies with the definition 2....solution is simple:

Adjust your register definition.

If you want the code to work differently, feel free to submit a pull request.

closed 2025-09-24 by janiversen

bambosd, 2025-09-24:
ok my friend... i'm just posting here because this function was including in the library and now my old code that was working, now is giving error.